kbdlab/walking_robot #25: sd카드에 문제가 생겼을 때

gud5201, 2019-11-15:
ssh를 연결을 위해 터미널에서 나노로 편집 중 해당 파일만 남고 우분투가 사라졌습니다.
 우선, 포맷을 하고 싶진 않아서 rescue tool들을 유튜브 등을 보고 따라해봤습니다만 잘 작동하지 않았습니다. 윈도우 cmd에서 ckdsk도 해봤는데 "파일 시스템 유형은 RAW입니다. RAW 드라이브에 대하여 CHKDSK을(를) 사용할 수 없습니다." 라는 메세지가 나오며 진단도 되지 않았습니다. Mac에서는 읽을 수 없는 디스크라고 나옵니다.
 이후 포맷을 해봤는데, 빠른 포맷은 안 되고, 느린 포맷이나, sd official에서 제공하는 format tool도 사용했는데 포맷이 되질 않습니다. etcher를 이용해서 ubuntu mate와 raspbian을 설치해봤는데 설치는 되는데 둘 다 작동은 안 됩니다.
 제가 할 수 있는 선에서 여러 방법을 시도해봤는데 해결이 되지 않아서  질문을 올려봅니다.
sd카드를 새로 사야할까요?

aegggsehyun1994, 2019-11-16:
윈도우, 맥과 마찬가지로 리눅스도 고유한 파일 시스템을 사용합니다. 윈도우, 맥은 오픈 소스가 아니다보니 자기 자신의 파일 시스템이 아니면 대체로 읽기가 안 되고요. 윈도우에서 RAW라고 뜬 거나, 맥에서 읽을 수 없는 디스크라고 나온 게 바로 무슨 파일 시스템인지 모르겠다는 뜻입니다. 앞으로 이런 일이 있으시면 리눅스에서 디스크를 읽어보세요. 리눅스는 파일 시스템이랑 관계 없이 어지간하면 모두 읽을 줄 압니다.

다만 포맷을 새로 했는데도 부팅이 안 되면 SD 카드 문제가 맞겠죠. 새로 구매하셔야 할 것 같네요. SD 카드가 사실 수명이 얼마 안 되니까 앞으로는 언제 SD 카드가 날아갈지 모른다는 생각으로 수업을 들으세요 ㅠㅠ 이건 어쩔 수 없네요.

gud5201, 2019-11-17:
답변 감사합니다.
말씀하신 대로 새로운 sd 카드를 사서 시도했는데 포맷 이후 인식이 안되고, 배드섹터가 생겼습니다. 따라서, 어댑터 문제일 수도 있다고 판단해 좀 더 좋은 어댑터로 교체하여 포맷팅 문제는 해결했습니다.(제공해주신 8GB sd카드는 죽었습니다ㅠ…)
 새로운 문제는, 라즈베리파이에 sd를 끼워도 부팅이 안됩니다. 초록빛이 깜빡여야 하는데 빨간 빛만 계속 들어와 있습니다. 이 경우엔 어떻게 해야 하나요?(검색 결과는 hardware failure라고 합니다.)

kbdlab, 2019-11-18:
처음부터 문제가 있다면 라즈베리 파이를 바꿔드리겠지만, 중간부터 문제가 발생한 경우 저희도 어쩔 수가 없습니다. 단, 연구실에 라즈베리 파이 여분은 있으니 새 라즈베리 파이 3를 구입해서 제공하는 조건으로 바로 라즈베리 파이 3 를 드릴 수는 있습니다.

gud5201, 2019-11-19:
오늘 연구실로 가면 되나요?

closed 2020-11-23 by aegggsehyun1994