kbdlab/walking_robot #28: 읽어온 이미지 파일의 last

kangsang1, 2019-11-29:
소켓통신 과정에서 if first != -1 and last != -1 : 
구문을 들어가지 못해서 first와 last를 출력해보니 
first=5, last =-1임을 확인했습니다. 
이러면 카메라가 사진을 찍지 못했다는 것인가요?

kbdlab, 2019-11-29:
-1은 발견하지 못했다는 의미입니다. 아래 코드처럼 잘 되어 있는지 잘 확인해주세요.

사진을 인코딩해서 보낸 다음에 읽는 과정에서 문제가 생긴 것이므로 사진을 찍지 못했다는 의미는 아닐 겁니다.

first = stream_bytes.find(b'\xff\xd8')
 last = stream_bytes.find(b'\xff\xd9')

kbdlab, 2019-12-02:
질문자는 이미 해결한것 같지만 저 원인에 대해 좀 더 알려드립니다. 기본적으로 소켓 통신은 서버(랩탑, 데스크탑 등)와 클라이언트(라즈베리 파이)가 통신을 하는 시스템이며, 이 때 클라이언트는 서버로 부터 데이터를 받아야 작동하게 되어 있습니다. 그래서 데이터를 받지 않으면 클라이언트는 정보를 보내주지 않습니다.

배포한 코드에서 서버가 클라이언트에게 주는 명령은 다음과 같습니다. order라는 string 형식의 파일을 인코딩해서 보내줍니다. 

> order=order.encode()
> self.request.sendall(order)

그런데 가끔 코드에서 이 명령이 다음 조건문 아래에 넣는 학생 분들이 많은 것 같습니다.

> if first != -1 and last != -1:

이 경우 조건문에 들어오지 않으면 서버는 클라이언트에 아무런 값을 주지 않기 때문에 처음 데이터를 한 번 전송하고 다시는 명령이 들어오지 않게 됩니다.

> stream_bytes += self.rfile.read(1024)

이 코드는 1024바이트만큼 데이터를 읽는다는 의미인데, 실제 데이터(320×240 컬러 이미지)는 대략 60000 바이트 정도 됩니다. 반복문이 60번은 실행되어야 할 만큼이죠. 따라서 첫번째 반복문 실행 시 first, 즉 이미지 파일의 처음 부분은 인식되었지만 서버에서 클라이언트로 보내는 명령이 실행되지 않으면 끝 부분은 영원히 받지 못하고 초기 상태만 반복적으로 나오게 됩니다.

이를 고쳐주기 위해서는 서버에서 클라이언트로 데이터를 계속 전송해주면 되는데, 꼭 의미 있는 데이터일 필요는 없으니 계속 아무거나 보내줘도 됩니다. 어차피 배포한 코드에서 모터에 넣을 출력은 m으로 시작하는 특정 문자열이니 모터에는 아무런 영향이 없습니다.

> self.request.sendall('1'.encode())

위와 같이 임의의 바이트를 보내면 됩니다.

다른 방법으로는 한번에 읽어 들이는 양을 늘리는 것인데, 너무 늘리면 오류가 발생합니다. 정확한 원인은 알 수 없으나 한 번에 너무 큰 양을 통신으로 하려니 문제가 발생하는 것 같습니다. 

아니면 원래 코드 처럼 if문에 들어가지 않은 상태에서 계속 통신을 해도 됩니다.

closed 2020-11-23 by aegggsehyun1994