kbdlab/walking_robot #34: Python 연습 과제 아이디어 공모

aegggsehyun1994, 2020-09-23:
황세현 조교입니다.

프로그래밍은 간단한 프로그램을 만들면서 배우는 게 가장 좋다 보니
연습을 위한 파이썬 프로그램 과제를 하나 내고자 합니다.

작년에는 술자리에서 하는 베스킨 라빈스 31 게임을 터미널에서 돌리는 프로그램을 만들었었는데요,
올해에는 행렬 연산, 이미지 처리 위주로 과제를 내고자 합니다.
현재 저는 프랙탈 그리기? 이런 걸 생각 중인데,
답글로 아이디어 주시면 과제에 고려해보도록 하겠습니다.

감사합니다.

ybkim95, 2020-09-23:
안녕하세요, 

아직 많이 부족하지만 시도해보면 재밌을 것 같은 주제를 몇 개 생각해 올려드립니다. 

1. 스트라센 알고리즘 구현해보기
2. V-REP api (python version)을 통해 V-REP 환경에서 카메라를 통해 이미지 인식 모듈 사용해보기
3. 개와 고양이를 구분하는 프로그램을 만들어보기
4. 개인 workspace를 terminal에서 구현해보기 (혹은 간단한 Linux file system 구현해보기)
5. 개인 웹 만들어보기
6. 많은 사람들 이미지 사이에서 본인을 찾아내는 프로그램 만들어보기
7. 윌리의 숨은 그림 찾기 책 사이에서 윌리 찾아보기 프로젝트
8. 사람의 사진을 input으로 받아 연령대와 성별 측정하는 프로그램 만들어보기
9. 본인보다 Atari 게임을 잘하는 에이전트를 학습시켜보기
10. yscec의 공지사항을 Web 크롤링해서 개인 Web에 표기해보기 

감사합니다.

aegggsehyun1994, 2020-09-24:
연습 과제는 얀센 로봇 프로그래밍에 직접적으로 쓰일 수 있는 범위에, 처음 파이썬을 배우는 사람도 한 2주 열심히 노력하면 해결할 수 있는 수준이었으면 해요. 4.를 할일 목록 정도로 단순화하고 7. 숨은 그림 찾기 정도는 취지에 맞을 거 같네요. 감사합니다.

blonix, 2020-09-24:
numpy 와 opencv를 활용해보는 정도면 괜찮을거 같습니다. numpy로 저장된 데이터를 불러와서 RGB 색 반전 결과물을 출력해본다던가..

closed 2020-11-23 by aegggsehyun1994