kbdlab/walking_robot #37: 프로젝트 관련 질문 드립니다.

kdi9694, 2020-10-02:
1. 제가 아래의 게시물의 지형크기 게시물을 보고 설정해보려다 사진의 frame 부분을 잘못 건드린 것 같은데, 시뮬레이션을 돌리면 5초 정도쯤에 갑자기 사라집니다. 프로그램을 삭제하고 다시 깔아봐도 해결이 안됩니다...
[image]

2. 시뮬레이션을 돌릴 때 저도 속도가 빨라 아래로 떨어지는데 어떻게 지형을 넓힐 수 있는지 궁금합니다

3. 혹시 컴퓨터 사양에 따라 시뮬레이션 결과가 차이나는지 궁금합니다. 같을 설정으로 노트북(샤양이 굉장히 좋지 않습니다.)이랑 데스크탑으로 시뮬레이션을 돌렸을 때, 데스크탑으로 48정도 나오는 속도가 노트북에서는 3정도 밖에 나오지 않습니다.

ywlee0305, 2020-10-03:
제가 지형을 변형한 방법은 이미 생성되어있는 지형의 일부분의 넓이를 키운것입니다.

아래 사진에 나와있는 순서대로 설정하고 실행하면 될것같습니다.

[image]

[image]

aegggsehyun1994, 2020-10-03:
1. 기본적인 거지만 씬을 지웠다가 다시 추가해보셨나요?
2. @ywlee0305 감사합니다.

dong9411, 2020-10-04:
3. 컴퓨터에 따라 속도가 다르게 나올 수 있습니다. 채점은 저희 연구실 컴퓨터 3대로 각각 돌려서 평균값으로 할 것 같습니다.

kdi9694, 2020-10-04:
감사합니다!! 잘 진행되고 있습니다!!

closed 2020-11-23 by aegggsehyun1994