kbdlab/walking_robot #41: TeamViewer

Reversable, 2020-11-04:
실습 시간에 알려주신 대로 

(모니터 없는 경우)
window powershell 통해서 라즈베리파이에 teamviewer 설치 하고
setup, login까지 했습니다.

그리고 나서 노트북에서 teamviewer 접속해 라즈베리파이 원격 제어를 하려고 하는데
[image]
첨부한 이미지와 같이 연결이 계속해서 실패합니다.

재부팅 후 시도해봤지만 계속해서 에러가 납니다.
어떻게 해야하나요?

dong9411, 2020-11-05:
혹시 메일 인증하셨나요?

Reversable, 2020-11-05:
넵 
[image]

이메일 인증도 했습니다.

dong9411, 2020-11-05:
sudo teamviewer license accept해보시고 안되면
sudo teamviewer info 스크린샷 찍어서 한번 보여주실래요?

Reversable, 2020-11-07:
license 는 accept 되었다고 떴습니다.

해결이 안돼서 image 재설치부터 몇 번 반복하니 해결됐습니다. 감사합니다.

closed 2020-11-23 by aegggsehyun1994