kbdlab/walking_robot #44: 이미지 스트리밍 질문입니다.

lcho3878, 2020-11-27:
같은 질문 죄송합니다. 두가지 여쭤보고 싶은 것이 있습니다.
ImageRW.py 에서

def Upload(body, headers={}):
    conn = HTTPConnection(host)
    conn.request('POST', '/', body=body, headers=headers)
    res = conn.getresponse()
    print(res.getheaders())
    print(res.getheader('X-Server2Client', 'Fallback'))
    print(res.read())
    print('Uploaded to', host, 'with status', res.status)

1. 이부분에서 body를 수정하라고 하셨는데 어떤 부분을 수정해야 body부분을 수정하는 것인지 잘 모르겠습니다.

2. 기울기를 이용한 방향결정 코드까지 짜논 상태이고 이를 decision.py로 저장해놓았는데
import decision을 이용하여 불필요한 코드를 줄이고 싶은데 1번 질문과는 별개로 import decision 을 하면 
'Incorrect buffer length for resolution %dx%d' % (width, height))
picamera.exc.PiCameraValueError: Incorrect buffer length for resolution 320x240
이라는 오류가 나면서 실행이 되지 않습니다.

aegggsehyun1994, 2020-11-27:
1. Upload(image) 이런 형식으로 저 Upload 함수를 사용하시면 됩니다. np.random의 결과를 적절한 형식으로 바꿔서 업로드에 넣는 코드가 UploadNumpy라는 이름으로 있으니 참고하세요.
2. 혹시 decision.py 파일 메일이나 깃허브 메시지로 보내주실 수 있나요?

lcho3878, 2020-11-27:
1. np.random의 결과를 바꾼다는 것은 
def UploadNumpy():
    img = 255 * np.random.random((100, 100, 3))
    print('shape', img.shape)
    result, img = cv2.imencode('.jpg', img, [int(cv2.IMWRITE_JPEG_QUALITY), 90])
    if not result:
        raise Exception('Image encode error')

    Upload(img.tobytes(), {
        "X-Client2Server" : "123"
    })
이 함수에서 img =  이부분을 수정하면 되는 건가요? 이부분을 수정해서 카메라로 촬영한 실시간 이미지를 받으려 하면 연결이 되지 않고 무한대기상태가 됩니다. 가장 기초적으로 보정도 하지 않은 실시간 이미지부터 시도해보려 깃허브에 있는 camera.py에 있는 코드를 가지고 image = frame.array 의 image를 이용해서
위에 있는 UploadNumpy() 함수에 img = image로 수정하여 코드를 실행시키면 무한대기상태가 되며 아무일도 일어나지 않습니다

2. 메일로 보냈습니다

aegggsehyun1994, 2020-11-27:
혹시 중간 중간에 프린트하는 코드 넣어서 어디서 멈추고 있는지 확인해보실 수 있나요?

lcho3878, 2020-11-27:
> 혹시 중간 중간에 프린트하는 코드 넣어서 어디서 멈추고 있는지 확인해보실 수 있나요?

확인해봤는데 img = image 받는 과정으로 넘어가지 못하는 것 같습니다.

def UploadNumpy():
    def select_white(image, white):
        lower = np.uint8([white,white,white])
        upper = np.uint8([255,255,255])
        white_mask = cv2.inRange(image, lower, upper)
        return white_mask

    camera = PiCamera()
    camera.resolution = (320, 240)
    camera.vflip = True
    camera.hflip = True
    camera.framerate = 30

    rawCapture = PiRGBArray(camera, size= (320, 240))

    time.sleep(.1)
    t = time.time()
    for frame in camera.capture_continuous(rawCapture, format="bgr", use_video_port=True):
        image = frame.array
        key = cv2.waitKey(1) & 0xFF
        rawCapture.truncate(0)

    img = image
    print('shape', img.shape)
    result, img = cv2.imencode('.jpg', img, [int(cv2.IMWRITE_JPEG_QUALITY), 90])
    if not result:
        raise Exception('Image encode error')
있는 코드를 참고하여 이렇게 작성했는데 이 부분에 문제가 있는 것 같습니다.

aegggsehyun1994, 2020-11-27:
`camera = PiCamera()
camera.resolution = (320, 240)
camera.vflip = True
camera.hflip = True
camera.framerate = 30

rawCapture = PiRGBArray(camera, size= (320, 240))

time.sleep(.1)
t = time.time()
for frame in camera.capture_continuous(rawCapture, format="bgr", use_video_port=True):
    image = frame.array
    key = cv2.waitKey(1) & 0xFF

    img = image
    print('shape', img.shape)
    result, img = cv2.imencode('.jpg', img, [int(cv2.IMWRITE_JPEG_QUALITY), 90])
    if not result:
        raise Exception('Image encode error')`
    #More image processing
    rawCapture.truncate(0)`

truncate를 옮기고 인덴트를 고쳤어요. More image processing 부분에 코딩을 하시면 됩니다.

lcho3878, 2020-11-27:
해결됐습니다 정말정말감사합니다 조교님!!!!

closed 2020-11-29 by aegggsehyun1994