kbdlab/walking_robot #46: image streaming legacy  질문입니다

ywlee0305, 2020-11-27:
server.py error:

Connection closed on thread 1
----------------------------------------
Exception happened during processing of request from ('192.168.0.5', 60410)
Traceback (most recent call last):
  File "/usr/lib/python3.6/socketserver.py", line 320, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 351, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 364, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.6/socketserver.py", line 724, in __init__
    self.handle()
  File "server_detect.py", line 89, in handle
    stream_bytes += self.rfile.read(1024)
AttributeError: 'KeyboardControl' object has no attribute 'rfile'

client.py  error:

Traceback (most recent call last):
  File "client_detection.py", line 36, in <module>
    connection.flush()
  File "/usr/lib/python3.7/socket.py", line 607, in write
    return self._sock.send(b)
BrokenPipeError: [Errno 32] Broken pipe

다음과 같이 에러가 나와 스트리밍이 안되는데 해결방법을 알 수 있을까요?

aegggsehyun1994, 2020-11-29:
원론적으로 답을 하면 self가 키보드 컨트롤 오브젝트네요. self가 BaseHTTPHandler로 나와야 rfile이 있을텐데요.

ywlee0305, 2020-11-29:
해결했습니다 감사합니다!

closed 2020-11-29 by aegggsehyun1994