kbdlab/walking_robot #48: 소켓통신을 통한 이미지 스트리밍 오류

SuhBoGyeong, 2020-11-28:
[image]

안녕하세요.
컴퓨터 wsl환경에서 먼저 Server.py를 실행시켜서 'HTTPServer Serving at port 8000'가 뜨고 난 뒤 라즈베리파이 ssh환경에서 ImageRW.py를 실행시켰는데 사진과 같이 ConnectionRefusedError: [Errno 111] Connection refused 가 뜹니다. 원인이 뭔지 알 수 있을까요??

aegggsehyun1994, 2020-11-29:
키보드 컨트롤 때는 잘 되셨나요? IP나 포트 다시 확인해보시고 파워셸에 `wsl -l -v` 쳐보세요.

dlwlrma71, 2020-11-29:
저도 마찬가지 오류가 뜹니다.. 
다만 저는 키보드 컨트롤 시 Client.py로만 실행 했을 때는 오류가 뜨고 Client.py 192.168...(노트북 ip)를 붙여주여야 제대로 실행되는데 원인이 무엇인지 알 수 있을까요?

aegggsehyun1994, 2020-11-29:
> 저도 마찬가지 오류가 뜹니다.. 
> 다만 저는 키보드 컨트롤 시 Client.py로만 실행 했을 때는 오류가 뜨고 Client.py 192.168...(노트북 ip)를 붙여주여야 제대로 실행되는데 원인이 무엇인지 알 수 있을까요?

네 원래 제가 코딩을 그렇게 한 겁니다 ㅋㅋㅋ

SuhBoGyeong, 2020-11-29:
네 키보드 컨트롤은 아래처럼 제대로 작동합니다ㅠ powershell에 쳤을 때 아래처럼 뜹니다.
[image]
[image]

aegggsehyun1994, 2020-11-29:
다들 ImageRW.py를 그대로 돌리고 계신 거 같네요. 그거랑 Client.py를 조합을 하는 코딩을 하셔야 합니다.

dlwlrma71, 2020-11-29:
넵 읽어보고 ImageRW.py 파일을 수정하니 해결된 것 같습니다. 감사합니다.

SuhBoGyeong, 2020-11-29:
오 코드 수정하니 연결되었습니다. 감사합니다!

closed 2020-11-29 by aegggsehyun1994