kbdlab/walking_robot #49: 어안렌즈 보정(calibrate.py)오류 질문있습니다

16Kim-Sol, 2020-11-28:
[image]

안녕하세요, 어안렌즈 보정을 위해 체크격자무늬 8장의 사진을 찍고 흑백으로 바꾸어 test_set3 폴더에 넣고, 위 사진과 같이 calibrate.py를 실행하니 오류가 생겼습니다. 어떻게 해야 오류 없이 실핼시킬수 있는지 알려주시면 감사하겠습니다.

[8p.zip](https://github.com/kbdlab/walking_robot/files/5610254/8p.zip)

test_set3에 있는 8장의 사진은 zip파일에 있는 jpg 8장입니다.

kdi9694, 2020-11-28:
[image]

저도 같은 오류가 떴었는데, calibration용 checkerboard 판 사진이 문제였던 것 같더라구요. 올려주신 사진 보면 완전 검정칸이랑 흰색 칸만 있는게 아니라 검은색 칸 테두리에 회색 선 같은게 있는거 같은데 제꺼도 그래서 인식을 못한다고 생각하고, 완전 검은색이랑 흰색으로만 되있는 사진으로 다시 뽑아서 찍으니까 됐어요!
구글에 검색할 때, checkerboard pattern calibration 까지 치면 잘 나오는거 같더라구요. 일단 저는 그렇게 해결을 했습니다.

16Kim-Sol, 2020-11-29:
감사합니다

closed 2020-11-29 by 16Kim-Sol