kbdlab/walking_robot #53: 소켓 통신관련

swho2, 2020-12-02:
그 새롭게 올려주신 Networking폴더의 Server.py 와 ImageRW.py를 이용하여 소켓통신을 하려 하였으나 잘 안되었습니다. 혹시 네트워킹이 먼저 잘 되지 않는 것같아 Server파일과 Client파일을 거의 수정하지 않은채로 연결을 시도해보니 다음과 같이 연결은 되어 Server쪽에서 input이 잘 써지는데 Client쪽에서는 input이 제대로 출력되지 않았습니다.
[image]

aegggsehyun1994, 2020-12-02:
몇가지 수정할 점 알려드릴게요.

1. 클라이언트에서 이미지 같은 Body를 보낼 때는 GET을 POST로 바꾸셔야 해요.
2. 현재 서버 코드에서는 키보드 입력을 받는 모드로 들어가면 라즈베리 파이가 이미지를 업로드하고자 해도 응답이 없습니다. 키보드를 누르지 않으시든지 GET 코드를 삭제하시든지 하셔야 할거에요. 키보드 입력도 동시에 보내고 싶으시면 [이 파일](https://github.com/kbdlab/walking_robot/blob/master/Networking/NonBlockingKeyInput.py)을 참조하세요.

swho2, 2020-12-02:
Server파일을 실행한 뒤에 ImageRW파일을 실행하였는데 다음과 같이 뜹니다.  해당 에러를 구글링하여 libxcb-xinerama0 패키지를 받아보았으나 에러가 잡히지 않았습니다.  그래서 그 2차때 했던 것처럼 키보드제어는 되나 하여 클라이언트파일을 실행해 보았으나 위의 질문에서 처럼 되지않았습니다.(클라이언트에서는 이미지를 보내는 것이 아니라 키보드 입력을 받았을때 모터속력이 어떻게 되는지 출력되는 본래의 코드에서 수정하지 않았습니다.)

[image]

[image]

aegggsehyun1994, 2020-12-03:
HTTP에 대해서 이해가 먼저 필요해보이네요. 클라이언트는 요청을 하고 서버는 이 요청에 응답을 전송하잖아요? 파이썬 특성 상 클라이언트에서 요청을 보냈는데 서버에서 아무 응답이 없으면 양쪽 코드는 그냥 정지 상태가 됩니다. GET과 POST 모두 이 원칙은 똑같고, 코드가 몇번째 줄에서 멈췄나 확인해가며 개발하시기를 조언드릴게요. 그리고 Could not connect to display 문제는 imshow를 빼시면 해결될거에요.

closed 2020-12-05 by aegggsehyun1994