kbdlab/walking_robot #54: Client, camera, ImageRW 코드 사용 문의

Reversable, 2020-12-05:
안녕하세요.
Client.py 에 camera.py 를 import해서 사용하려고 합니다.
그런데 아래 코드를 실행시키게 되면 connected refused가 뜹니다.
혹시 어디가 문제인지 알 수 있을까요?
코드에 대한 설명은 아래에 작성했습니다!

<Client.py>
PORT = 8000

from http.client import HTTPConnection
import json
from time import sleep
import numpy as np
import cv2

from Time import Time
from sys import argv

print(argv)

HALF=.5
MOTOR_SPEEDS = {
    "q": (HALF, 1), "w": (1, 1), "e": (1, HALF),
    "a": (-1, 1), "s": (0, 0), "d": (1, -1),
    "z": (-HALF, -1), "x": (-1, -1), "c": (1, HALF),
}

def main():
    while True:
        conn = HTTPConnection(f"{argv[1] if len(argv) > 1 else  'localhost'}:{PORT}")

        try:
            conn.request("POST", "/")
        except ConnectionRefusedError as error:
            print(error)
            sleep(1)
            continue

        print('Connected')
        res = conn.getresponse()

        **import Camera**
        
        while True:
            chunk = res.readlines()
            if (chunk == b'\n'): continue
            if (not chunk): break

            chunk = chunk[:-1].decode()
            data = json.loads(chunk)
            print(Time(), data)
            action = data['action']
            print('action', action)
            try:
                print(MOTOR_SPEEDS[action])
            except KeyError as error:
                print(error)

main()

-----------

위 코드처럼 Client.py에 Camera.py를 import해서 사용했습니다.
Camera.py에는 ImageRW의 함수들을 함께 내장시켰습니다.

<Camera.py>
(import, ImageRW의 함수들 제외한 코드입니다.)

camera = PiCamera()
camera.resolution = (320, 240)
camera.vflip = True
camera.hflip = True
camera.framerate = 30

rawCapture = PiRGBArray(camera, size= (320, 240))

time.sleep(.1)
t = time.time()

for frame in camera.capture_continuous(rawCapture, format="bgr", use_video_port=True):
    image = frame.array
    mask_image = select_white(image, 160)

    gray = cv2.cvtColor(image,cv2.COLOR_BGR2GRAY)
    key = cv2.waitKey(1) & 0xFF

#UploadNumpy 적용
    img = mask_image
    print('shape', img.shape)
    result, img = cv2.imencode('.jpg', img, [int(cv2.IMWRITE_JPEG_QUALITY), 90])
    if not result:
        raise Exception('Image encode error')

    rawCapture.truncate(0)		

    Upload(mask_image)

    if key == ord('q'):		
        break

aegggsehyun1994, 2020-12-06:
일단 지우셔야 하는 게 너무 많이 보이네요 ㅠ `for frame in camera.capture_continuous`를 제외한 `while`은 모두 필요가 없습니다. 그리고 POST할 때 body로 이미지를 보내는 부분을 빼먹으셨고, 2차 때와 다른 점이 이번에는 이미지를 보낼 때마다 연결을 새로 수립한다는 점인데, `res.read()`를 시작으로 그 이후를 생각해보시기 바랍니다.

Connection refused는 IP나 포트가 잘못됐다고밖에는 볼 수 없습니다. argv를 빼먹은 건 아니신가요.

Reversable, 2020-12-06:
POST할 때 body로 이미지를 보내는 부분은 아래와 같이 수정하면 되는 건가요?
(Upload() 함수를 이용해서 POST 할 생각입니다.)

    def Upload(body, headers={}):
        conn = HTTPConnection(host)
        conn.request('POST', '/', body=body, headers=headers)
        res = conn.getresponse()
        print(res.getheaders())
        print(res.getheader('X-Server2Client', 'Fallback'))
        print(res.read())
        print('Uploaded to', host, 'with status', res.status)

aegggsehyun1994, 2020-12-06:
네 body 형식에만 유의해주세요. jpg로 바꾸는 거 잊지 마시고 바이트 형식으로 보내야 해요.

Reversable, 2020-12-07:
네 감사합니다!

closed 2020-12-07 by Reversable